wokwi/wokwi-features #927: WiFi for ESP32-C6 is not working.

Koepel, 2024-12-06:
It was mentioned in the Discord channel that the WiFi for the ESP32-C6 is not working.
I took the basic example from the documentation here: https://docs.wokwi.com/guides/esp32-wifi#connecting-from-arduino
Test project: https://wokwi.com/projects/416535802836071425
It shows "error: pll_cal exceeds 2ms!!!".

gregor-coder, 2024-12-08:
I also ran into this issue. Please help in solving this issue.

urish, 2024-12-09:
Issue reproduced, looking into it.

urish, 2024-12-09:
"error: pll_cal exceeds 2ms!!!" resolved - unfortunately, it still won't connect.

urish, 2025-02-11:
Fixed. This was a tricky one - it wasn't a bug in the simulator, rather in the JavaScript tooling that we use for production (swc, part of Next.js)