wokwi/wokwi-features #930: DMA doesn't work with ILI9341

orsinium, 2024-12-09:
**Describe the bug**

I'm trying to run [rust-esp32s3-ili9341](https://github.com/bjoernQ/rust-esp32s3-ili9341) project in wokwi. It uses DMA for working with the display and it is supposed to work on the real hardware. However, it freezes when running it in wokwi.

**To Reproduce**

1. `git clone git@github.com:bjoernQ/rust-esp32s3-ili9341.git`
2. `cd rust-esp32s3-ili9341`
3. `cargo build`
4. add files listed below
5. run the simulation

wokwi.toml:

```toml
[wokwi]
version = 1
gdbServerPort = 3333
elf = "target/xtensa-esp32s3-none-elf/debug/ili9431_esp32s3_test"
firmware = "target/xtensa-esp32s3-none-elf/debug/ili9431_esp32s3_test"
```

diagram.json:

```json
{
  "version": 1,
  "editor": "wokwi",
  "author": "orsinium",
  "parts": [
    {
      "type": "board-esp32-s3-devkitc-1",
      "id": "esp",
      "top": 136.38,
      "left": -1.28,
      "rotate": 270,
      "attrs": {}
    },
    {
      "type": "wokwi-ili9341",
      "id": "lcd1",
      "top": -55.61,
      "left": -38.21,
      "rotate": 270,
      "attrs": {}
    }
  ],
  "connections": [
    ["esp:TX", "$serialMonitor:RX", "", []],
    ["esp:RX", "$serialMonitor:TX", "", []],
    ["lcd1:SCK", "esp:7", "orange", ["h41.59", "v128.76", "h-180.28"]],
    ["lcd1:MOSI", "esp:6", "blue", ["h35.45", "v124.53", "h-164.54"]],
    ["lcd1:MISO", "esp:2", "blue", ["h35.45", "v124.53", "h-164.54"]],
    ["esp:GND.1", "lcd1:GND", "black", ["v12.64", "h65.1", "v-200.45"]],
    ["lcd1:CS", "esp:5", "black", ["h21.64", "v9.6"]],
    ["esp:4", "lcd1:D/C", "green", ["v24.15", "h244.8", "v-240.76"]],
    ["lcd1:VCC", "esp:5V", "red", ["h16.27", "v196.99", "h-65.22"]],
    ["esp:3V3.1", "lcd1:LED", "purple", ["h17.27", "v197.99", "h-67.22"]],
    ["esp:48", "lcd1:RST", "blue", ["h17.27", "v197.99", "h-67.22"]]
  ],
  "serialMonitor": {
    "display": "terminal"
  }
}
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**

 - OS: Linux
 - Browser: vscode extension

**Additional context**
Add any other context about the problem here.

Original issue: https://github.com/bjoernQ/rust-esp32s3-ili9341/issues/2

urish, 2024-12-09:
Pushed a fix to DMA on ESP32-S3 - can you please test again now?

orsinium, 2024-12-09:
Still doesn't work. I've tried both vscode extension and the online playground.

orsinium, 2024-12-09:
Oh, wait, it works if I build the release version: `cargo build --release` instead of `cargo build` 🤔

orsinium, 2024-12-09:
Alright, that's good enough for me. I use release build for my projects anyway. Thank you :) I had another issue with SD card. I tried to ask in Discord but no luck. I'll open a separate issue later 👍🏿

closed 2024-12-09 by orsinium