wokwi/wokwi-features #934: TM1637 decimal point implementation

NewAbbreviations950, 2024-12-11:
Would it be possible to implement functionality for the decimal points on the tm1637 4 digit display? They appear in the picture of the display but don’t seem to be implemented in the emulator.

djedu28, 2024-12-14:
This flaw was commented on:
https://github.com/wokwi/wokwi-features/issues/279#issuecomment-1761570712

NewAbbreviations950, 2024-12-14:
I see, but it's not fixed. I'm using the Pico SDK and the bit corresponding to the dp segment doesn't do anything. So when I send a byte the 7 segments work as expected but the dp doesn't.

urish, 2024-12-14:
Hi, can you please link to a Wokwi project that reproduces the issue?

urish, 2024-12-14:
The current TM1637 part emulates the following module:

[image]

I looked up and see there's another variant, which has decimal points instead of colons.

NewAbbreviations950, 2024-12-14:
Ah ok, the picture while you're using it is the variant with decimal points. But that explains it.
Would it be possible to add the variant with the decimal points then?

urish, 2026-10-09:
The decimal point module variant is now implemented. To use it, add `"type": "dp"` to the attrs of the TM1637 part in your diagram.json:

```json
{ "type": "wokwi-tm1637-7segment", "id": "display1", "attrs": { "type": "dp" } }
```

With this variant, bit 7 of each digit byte lights that digit's decimal point instead of the colon. Details: https://docs.wokwi.com/parts/wokwi-tm1637-7segment#attributes

closed 2026-10-09 by urish