wokwi/wokwi-features #935: Move Runtime/Cpu Meters a little to the left in the VS Code extension

HerbieJay, 2024-12-18:
When running the simulator in VS Code the runtime/cpu percentage meter are all the way to the right of the frame. When the cpu meter goes from 2 digits (99) to 3 (anything 100 and larger) the frame produces scroll bars for in both the horzinotal and vertical directions. When the simulation keeps running if the cpu percentage drops below 100 the scroll bars disappear. This causes the screen to flicker and is very distracting when watching the simulator run. I am hoping that the information readout can be nudged to the left by a few pixels and hopefully remote this issue from my setup. I am running VS Code on an Macbook Air. Here is the version info incase it maters

Version: 1.96.0 (Universal)
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z (1 wk ago)
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Darwin arm64 24.2.0

urish, 2024-12-20:
Thanks for reporting - I'm trying to reproduce this behavior, but without luck so far.

Do you get a similar behavior in the online version of wokwi, e.g. https://wokwi.com/projects/new/esp32 ?

HerbieJay, 2024-12-20:
I do not get it using the online editor or the VS Code extension on a windows 11 box. Just on my MacBook Air.

bobziuchkovski, 2024-12-21:
I'm also seeing this on my end. MacOS 14.7 with VSCode 1.96.0 and Wokwi extension 2.6.0. Here's what it looks like: https://youtu.be/NNVVo2RVlus

HerbieJay, 2024-12-21:
Yes! that is exactly what I am seeing on my Mac.

bobziuchkovski, 2024-12-22:
My VSCode font config in case that's also playing a role in this:

```
    "editor.fontFamily": "Menlo, Consolas, DejaVu Sans Mono, monospace",
    "editor.fontSize": 14,
```

urish, 2024-12-26:
Should be fixed now! Can you please test and report?

bobziuchkovski, 2024-12-26:
Confirmed! I no longer see the scrollbars flash on and off. Thanks!

closed 2024-12-26 by urish