wokwi/wokwi-features #936: Request for Adding ESP 01 Module

amithprasannas, 2024-12-19:
ESP 01 module is a multipurpose module that we can use to implement OTA features on other DEV boards/chips that have no direct Wifi or Wireless connections to the internet.

djedu28, 2024-12-25:
esp01 as a wifi module is already supported on wokwi

https://wokwi.com/projects/393358172882383873
[image]

to use, just insert `{ "type": "board-esp-01", "id": "esp01", "attrs": {} }` in `diagram.json->parts`

## diagram.json
```diff
{
  "version": 1,
  "author": "Anonymous maker",
  "editor": "wokwi",
  "parts": [
    ...
+    { "type": "board-esp-01", "id": "esp01", "attrs": {} }
  ],
  "connections": [
   ...
  ],
  ...
}
```

djedu28, 2024-12-25:
It's not documented, but it's possible to find Wokwi projects with `board-esp-01` on Google https://www.google.com/search?q=esp01+wokwi

---- 

Any help is welcome to document `board-esp-01` in Wokwi.
* Documentation for the Wokwi Simulator  - https://github.com/wokwi/wokwi-docs