wokwi/wokwi-features #938: DCF77 feature

plainprince, 2024-12-23:
For now, there is only a RTC module for getting the time. RTC modules use quartz's so they are not really exact. The DCF77 syncs to atomic clocks, and in most finished products, is of course used more often than RTC modules since they are and will stay perfectly on time.

djedu28, 2025-01-01:
> For now, there is only a RTC module for getting the time. 

it's not entirely true. You can use an ESP32, RP or Arduino+ESP01 (as a wifi module) and get the time over the internet using NTP

> RTC modules use quartz's so they are not really exact.

RTC Accuracy:

- Most RTCs have an accuracy of ±1 minute per year, which is acceptable for many applications.
- For applications that require high time accuracy, it is recommended to use an external RTC with a high-frequency oscillator, or a GPS or NTP for continuous synchronization.
- In home or educational projects, the accuracy of the RTC is usually sufficient.

> The DCF77 syncs to atomic clocks, and in most finished products, is of course used more often than RTC modules since they are and will stay perfectly on time.

1 - The official DCF77 signal works in Germany and most of Europe. The DCF77 transmitter is located in Frankfurt/Mainflingen and distributes the signal over long radio waves with a frequency of 77.5 kHz, within a radius of 1500 km. 
- In other words, it is not an accessible technology for all wokwi users (spread across the globe)
- On the other hand, RTC works in any region of the globe, being accessible to all wokwi users.

2 - In fact, most computer networks use NTP to distribute the DCF77 time signal. NTP servers can receive DCF, MSF, and WVBB, as well as many of them can also receive the GPS signal. 

----
automatically translated
<details><summary>Original text in pt-BR</summary>
> For now, there is only a RTC module for getting the time. 

Não é totalmente verdade. Você pode utilizar um ESP32, RP ou Arduino+ESP01 (como modulo wifi) e obter a hora pela internet usando o NTP.

> RTC modules use quartz's so they are not really exact.

Precisão do RTC:

- A maioria dos RTCs tem uma precisão de ±1 minuto por ano, o que é aceitável para muitas aplicações.
- Para aplicações que requerem alta precisão de tempo, é recomendável usar um RTC externo com um oscilador de alta frequência, ou um GPS/NTP para sincronização contínua.
- Em projetos domésticos ou educacionais, a precisão do RTC geralmente é suficiente.

> The DCF77 syncs to atomic clocks, and in most finished products, is of course used more often than RTC modules since they are and will stay perfectly on time.

1 - O sinal DCF77 oficial funciona em Alemanha e em grande parte da Europa. O transmissor DCF77 está localizado em Frankfurt/Mainflingen e distribui o sinal por ondas de rádio longas com uma frequência de 77,5 kHz, dentro de um raio de 1500 km. 
- Ou seja, não é uma tecnologia acessível para todos os utilizadores do wokwi (espalhados pelo globo)
- Em contrapartida, o RTC funciona em qualquer região do globo, sendo acessível para todos os utilizadores do wokwi.

2 - Na verdade, a maioria das redes de computadores usa NTP para distribuir o sinal de tempo DCF77. Os servidores NTP podem receber DCF, MSF e WVBB, bem como muitos deles também podem receber o sinal GPS. 

</details>

djedu28, 2025-01-01:
You can build the DCF77 receiver simulation using the custom chips api in wokwi.

* https://docs.wokwi.com/chips-api/getting-started

---- 

Automatically translated.

<details>

<summary>ordinal text in Brazilian Portuguese</summary>
<br>
Você pode construir a simulação do receptor de DCF77 usando a api de chips customizados no wokwi.

* https://docs.wokwi.com/chips-api/getting-started

</details>