wokwi/wokwi-features #940: more speakers and motor options plsss

DimiGames, 2024-12-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

djedu28, 2025-01-01:
Please describe.

1. Read the description of the issue you created. There is an example framework that you should base yourself on.

> ## Is your feature request related to a problem? Please describe.
> A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
> 
> ## Describe the solution you'd like
> A clear and concise description of what you want to happen.

2. Simply placing generic text in the title is not relevant, and it will be closed due to lack of information.

urish, 2025-01-04:
Closing as the request is missing specific details.

closed 2025-01-04 by urish