wokwi/wokwi-features #942: Unsaveable anonymous project oddness

alto777, 2025-01-04:
Adding a line with a certain URL makes anonymous projects unsaveable

Add this line

  Serial.println("\nhttps://forum.arduino.cc/t/cloud-platform-and-ide-together/1337359\n");

to the setup() function ofthis project

https://wokwi.com/projects/419159493715571713

and it will no longer save.

So far it happens if the URL is a comment or in a print statement literally. I have tried two URLs, bith from Arduino fora.

A clear and concise description of what you expected to happen is that adding one line like this should not cause an anonymous project to go unsaveable.

Macos SoNoMa 14.7.1
Chrome 131.0.6778.205

I started noticing a save problem and finally found it to depend on including, in various ways, the URL.

I tried

```
  Serial.print("\nh tt s://forum.arduino.cc/t/c");
  Serial.println("loud-p latform-and-i de-togethe r/1337359\n");
```

and failed, sprinkling in some spaces makes it work

```
  Serial.print("\nh tt s://forum.arduino.cc/t/c");
  Serial.println("loud-p latform-and-i de-togethe r/1337359\n");
```

An explanation would be almost as good. I don't even begin to wonder how this is happening other than by design.

a7

sutaburosu, 2025-01-04:
> I don't even begin to wonder how this is happening other than by design.

I imagine this is due to the huge number of spam projects that were discovered using wokwi for link farming. Unless this is a very recent change, I don't think it affects me, so there may also be a list of users permitted to include URLs in their sketches.

urish, 2025-01-04:
Yes, this is related to the SPAM filtering. You need to be signed in if you want to save projects that contain external links.

closed 2025-01-04 by urish