wokwi/wokwi-features #945: Can't open sketch: no such file or directory: /sketch

plainprince, 2025-01-09:
**Describe the bug**
I started making a little project in the simulator, and today I started, clicked the run button and it told me "Can't open sketch: no such file or directory: /sketch"

**To Reproduce**
I have no idea how to reproduce this, here is a possible thing:
- Use the weirdest browser you have
- make a little project with a ssd1306 OLED and an analog joystick
- put your computer into hybernation mode (save RAM onto disk)
- wait
- maybe it'll break.

**Expected behavior**
The simulation should start running.

**Environment (please complete the following information):**
 - OS: MacOS
 - Browser: old version of chrome
 - Version: I think 111

**Additional context**
Note: this is an old chrome/chromium version. It might be breaking because the version is too old, but it definitely shouldn't, since compiling runs on the server.

plainprince, 2025-01-09:
Oh - my project is https://wokwi.com/projects/419530747484713985

plainprince, 2025-01-09:
In the meantime, I'm gonna re-create/duplicate my project.

sutaburosu, 2025-01-09:
I think was a temporary issue with the build server. Your project compiles and works fine for me after adding a blank line to your project (to invalidate the cached build result).

plainprince, 2025-01-09:
It still doesn't seem to work. I tried reloading and saving.

plainprince, 2025-01-09:
Oh also can you automatically add a newline to the end of the file in the compiler (as a fix)? Thanks a lot.