wokwi/wokwi-features #946: Compiles other project

plainprince, 2025-01-09:
**Describe the bug**
I kept playing around with my project, and then it started connecting to wifi (which it shouldn't. It's nowhere in the  code). after changing a semicolon in my code for recompilation, it started working again.

**To Reproduce**
Just compile a million billion times and hope it'll glitch.

**Expected behavior**
The correct project should start running

**Environment (please complete the following information):**
 - OS: MacOS
 - Browser Chrome
 - Version 111

**Additional context**
Yes I know this is the second issue today.
I'm pretty sure the server mixed up two compiles.

urish, 2025-01-10:
Unfortunately, without a way to reproduce, there's not much we can do about it. A screen recording showing this happening might help.

plainprince, 2025-01-10:
Well sadly I don't 24/7 screen record and the probabilities of this happening are very low. I think the compile ID or something might be something like new Date() or Date.now() (get the current UTC time in ms) so if two people compile and it happens in the exact same moment it might happen. My idea to reproduce would be to use something like puppeteer to open two browsers and compile. then try a few times and both might be mixed up, though the probabilities will still be very low so it's probably not a good way to reproduce, and you'll have to try like a million times.

urish, 2025-01-11:
Thanks for explaining, will investigate.

urish, 2025-01-19:
Thanks again for reporting - I believe we found and fixed the issue. It happened in some cases where there was a build error (such as specifying an invalid library in libraries.txt).