wokwi/wokwi-features #947: VSCode WokWi Diagram Editor does not understand custom chips, dropping connections to them and labelling them as "Missing"

pekkanikander, 2025-01-14:
**Describe the bug**

When running the WokWi Diagram Editor as the editor in an editor tab in VSCode, for `diagram.json`, if there is a custom chip in the project, e.g. `custom.chip.json` and `custom.chip.c` files, and the `diagram.json` is referring to this in `parts` as `"type": "chip-custom"`, the WokWi Diagram Editor is not able to upload the `custom.chip.json` file from the host-local VSCode folder but instead shows a confusing error message `Missing chip chip-custom` in the part image in the editor. 

**To Reproduce**

1. Install VSCode and WokWi Simulator extension to VSCode (if you don't have them already)
2. Clone any suitable WokWi GIT repository, with a custom chip, to VSCode, e.g. https://github.com/pekkanikander/Test2 
3. Open `diagram.json` for editing
4. See the WokWi Diagram Editor to appear.

**Expected behavior**

The custom chip in the project (STM1VAFE3BX in `Test2`) should appear as the defined custom chip, with connections to the other chips.

**Current behavior**

The custom chip in the project (STM1VAFE3BX in `Test2`) appears as a `Missing chip`, with no connections to other chips.

**Environment**
 
macOS 15.2 (24C101)
Safari 18.2 (20620.1.16.11.8)
VSCodium 1.96.2 Release: 24355, 
WokWi Simulator extension 2.6.0 - 2024-12-10.

**Additional context**

At https://wokwi.com/_next/static/chunks/4729-9dd899e2e960fade.js#L19795 in function `getChip(t)`, the chips are only looked at  `https://wokwi.com/api/external/chips`.  This does not make sense here, since the chip definition is on the local disk of the VScode host. For custom chips defined in the project, the WokWi extension should first upload the local file to somewhere in the `wokwi.com` domain so that the WokWi Editor, running in the iframe, could download and read it. The WokWi Editor cannot try to read the file directly from the local disk, as the iframe cannot escape its sandbox.  Or, if the WokWi Simulator extension already does upload the file for including it in `https://wokwi.com/api/external/chips`, there appears to be a bug somewhere else.

As a workaround, one could add a possibility to define a custom URL path for custom chips, allowing the WokWi Editor in the iframe to load the code from a custom URL (if allowed by sandboxing, I don't remember...).

( <rant> I hate debugging closed source obfuscated JS code. Hence, there may be errors in my analysis. I didn't want to spend too much time in debugging this. </rant> )

mmxavier, 2025-01-30:
I`m facing the same problem with no evolution in solutions.

hhhouran, 2025-02-04:
Also facing this issue as of now. This is the only issue found as of now. Any updates?

oscar88leonardo, 2026-10-06:
Hi, i am having the same issue, using platformio in cursor/vscode.

 i have compiled the .wasm file, but  in the diagram Editor i get "Missing chip" and no connections.

my wokwi.toml 

[[chip]]
name='bmp280'
binary = 'bmp280.chip.wasm'

[wokwi]
version = 1
firmware = '.pio/build/upesy_wroom/firmware.bin'
elf = '.pio/build/upesy_wroom/firmware.elf'

i have in the same root directory for the project the files: wokwi.toml, wokwi-api.h, platformio.ini, diagram.json, bmp280.chip.wasm, bmp280.chip.json, bmp280.chip.c. 

i am also with the community license.

thanks in advance

urish, 2026-10-06:
First, two updates my previous comment report:

1. You no longer need a dist folder or a manual toolchain: `wokwi-cli chip compile bmp280.chip.c` produces a bmp280.chip.wasm next to the source. See: https://docs.wokwi.com/guides/custom-chips-to-wasm.
2. The diagram editor does load local chips from the [[chip]] entries in wokwi.toml, but it reads them only once, when the editor tab opens, and it currently needs the .wasm to exist even though it only uses the .chip.json to draw the part.

So for the "Missing chip" symptom, please check:

1. wokwi.toml is in the same folder as diagram.json.
2. The .chip.json sits next to the .chip.wasm with the same base name (bmp280.chip.wasm and bmp280.chip.json).
3. Close and reopen the diagram.json tab after compiling the chip or editing wokwi.toml. Any error from loading the chip would show as a VS Code notification at that moment.

@oscar88leonardo, your layout looks right, so step 3 is my best guess for you. Could you confirm?