wokwi/wokwi-features #948: core-s3 - freeze ?

tcpipchip, 2025-01-16:
https://wokwi.com/projects/402847860051798017

**Describe the bug**
No run. 

**To Reproduce**
Just play the run button

**Expected behavior**
Freeze, nothing showed in LCD

**Environment (please complete the following information):**
 - WINDOWS 10
 - IE
 - Last

**Additional context**
Nothing

tcpipchip, 2025-01-16:
btw, if you upload the BIN, works

tcpipchip, 2025-01-16:
if upload a bin, works

[image]

urish, 2025-01-18:
This can be fixed by configuring the serial interface to USB_SERIAL_JTAG:

```json
{
  "version": 1,
  "author": "Ambitious maker",
  "editor": "wokwi",
  "parts": [
    {
      "type": "board-m5stack-core-s3",
      "id": "board",
      "top": -179.23,
      "left": -62.59,
      "attrs": { "serialInterface": "USB_SERIAL_JTAG" }
    }
  ],
  "connections": [ [ "$serialMonitor:RX", "board:G43", "", [] ], [ "$serialMonitor:TX", "board:G44", "", [] ] ],
  "dependencies": {}
}
```

We'll fix the example accordingly

tcpipchip, 2025-01-18:
Team,
Is it possible to open this web project, passing as parameters the path of bin/hex to execute ?
We have a nocode app that has a web browser client. This nocode app creates a bin/hex. We want to call this url passing the path of bin without use the UPLOAD CODE AND EXECUTE.
We have our webbrowser inside the app. Do you understand?

urish, 2025-01-19:
Example fixed.

We have an experimental support for sending the bin / diagram via query params, see https://wokwi.com/experimental/viewer

closed 2025-01-19 by urish

tcpipchip, 2025-01-19:
Wow!!!!

tcpipchip, 2025-01-19:
We tried (me and more two friends).
No works.
Can you send us an example with two parameters ?

urish, 2025-01-19:
Here is an example: https://wokwi.com/experimental/viewer?diagram=https%3A%2F%2Fraw.githubusercontent.com%2Fgeorgik%2Fesp32-sdl3-example%2Fmain%2Fboards%2Fesp32_p4_function_ev_board%2Fdiagram.json&firmware=https%3A%2F%2Fgithub.com%2Fgeorgik%2Fesp32-sdl3-example%2Freleases%2Fdownload%2Fv0.2.0%2Fesp32-sdl3-example-esp32_p4_function_ev_board.bin