wokwi/wokwi-features #949: Arduino nano Dallas DS18B20 output wrong temperature

ren4x, 2025-01-17:
Helo, I am testing temperature sensor with arduino nano. I slide temperature slider to maximum 125°C but get output : 15.69 °C

there is public sample project: 
https://wokwi.com/projects/420320796362385409

github-ec, 2025-01-18:
I second this observation.

While example code works like 

[https://wokwi.com/projects/358129234661487617](url)

it provides wrong values after adding an empty line e.g. between the includes and the variable declarations. 

As this seems to be true for different examples it may be related to  (re-)compilation of the sketches. I assume that the simulator uses an (older) compiled version if started without any change ...

urish, 2025-01-18:
Issue reproduced, thanks for reporting!

urish, 2025-01-19:
Looks like this is related to a new release of the DallasTemperature library.

As a workaround, use `DallasTemperature@3.9.0` in your libraries.json: https://wokwi.com/projects/420521250626544641