wokwi/wokwi-features #950: DS18B20 wrong temperature and other problems

Koepel, 2025-01-18:
Is this the right place to report a problem with the simulation of the DS18B20 ?
Problems with the DS18B20 was reported in the Discord channel.

1. Device ID. The keyword in diagram.json has changed from "device_id" to "deviceID".
2. Temperature. The returned temperature is wrong. It worked in the past.

During testing, there are compatibility problems with other boards and some functions of the OneWire library. I can not pinpoint those, so for now it is about the Arduino Uno with the DallasTemperature library and multiple DS18B20 sensors.

Test project: https://wokwi.com/projects/420427809886873601

Koepel, 2025-01-18:
I just noticed that the previous Issue is the same: https://github.com/wokwi/wokwi-features/issues/949

urish, 2025-01-19:
Closing (duplicate of #949)

closed 2025-01-19 by urish