wokwi/wokwi-features #959: wokwi vscode extension not working

rafaelcorsi, 2025-02-20:
Hello, 

Since today, the vscode extension stopped working. The behavior is a blank screen when opening a diagram.json file and when I try to run any wokwi vscode commands, i receive `command wokwi-vscode.start not found`

This happens to me and with several other students, in both linux and windows machines.

[image]

[image]

all the best,

Rafael

urish, 2025-02-20:
Thanks for reporting, looking into this! 

For now, a quick workaround would be to upgrade to version 2.6.0

urish, 2025-02-20:
[image]

closed 2025-02-20 by urish