wokwi/wokwi-features #962: The My Projects personal page stopped generating and showing the miniatures

GabyGold67, 2025-02-24:
**Describe the bug**
For some time the My Projects personal page was "sluggish" or took some time to generate the miniature depicting the project, now it seems completely stuck, not generating the miniatures ever, in my case for the last 3 days. @Koepel confirmed the problem.

**To Reproduce**
Generate a new project, save it, wait the miniature to be generated. I just tryed a simple one, justa an Arduino Nano and a led... it waited and failed.

**Expected behavior**
Have the miniature generated...

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome
 - Version 133.0.6943.127 

**Additional context**
In the already simplistic **My Projects** page where the resources available to sort and find a project, in a ever growing collection  gets complicated, the miniatures gives a visual hint to find the  one we're looking for. And that's why I consider it needed.

Constructing comment: 
Having the two columns layout for the projects makes the readable section of the project name short, so  imporvising a folder tree to clasify them like I do (**environment/project/library/demoFileName**) becomes useless, there's no sorting capabilities and now there are no miniatures.

urish, 2025-02-24:
Thanks for reporting! I've allocated more resources to the service that generates the thumbnails. Can you please check again now?

GabyGold67, 2025-02-24:
Hi Uri,
   No, not generated... And if this adds some information at all, when I
open the project after a few seconds I see a pop-up message, photo
attached. Stranger as I'm already a paying subscriber to a Hobbyist+ plan...
Hope this helps.
Gaby.//

urish, 2025-02-24:
Thanks for testing! 

Can't see the photo - you might need to go to the GitHub webpage to attach it properly.

GabyGold67, 2025-02-24:
Sorry about that, I'll be sending it in a few minutes

urish, 2025-02-24:
Thanks! Are you getting this for a specific project, or for multiple projects?
Can you share the link?

Also, which browser are you using?)

(I can't reproduce it here, and I also checked the build server logs, and the build servers for paid users are not under heavy load or anything).

urish, 2025-02-24:
> It was appearing randomly on those last projects, after I entered into them, like if it was set asked to run as soon as I entered automatically.

Ok, thanks! Now I know what's happening. When you enter a project, Wokwi automatically tries to compile it, to make sure it's ready sooner if you want to run it. However, in that case, it's always using the standard build servers, not the premium ones. It should not display the warning if that fails. I'll fix that.

GabyGold67, 2025-02-24:
Ah, OK, that makes sense, and it seems you could get the real problem out of the nonsense fluffy mishmash description I made.

Thank you!

Gaby.//

urish, 2025-02-24:
Sure thing - and now, it should be fixed too (if you didn't request a build, you should not see an error). 

Please do let me know if this issue ever repeats!

closed 2025-02-24 by urish