wokwi/wokwi-features #964: Add interrupt pin functionality to MPU6050

ndpegram, 2025-03-01:
Currently, the MPU6050's interrupt pin is not functional.

I wish to test using this pin in my scripts, so it would be appreciated if it were enabled in Wokwi.

urish, 2025-03-02:
Hi, can you please provide a link to a Wokwi project with the relevant code?

ndpegram, 2025-03-07:
I haven't attempted this in a project, as the information in the wiki at https://docs.wokwi.com/parts/wokwi-mpu6050 says that the pin is not functional and to request them if you need that functionality. Hence my request.

urish, 2025-03-11:
Thanks for explaining! 

Including a link to a relevant test project that have been tested on physical hardware will increase the chances we'll get to implementing it sooner.