wokwi/wokwi-features #966: wokwi-tm1637-7segment part broken?

maxint-rd, 2025-03-03:
Hmmm... as of today it seems that the wokwi-tm1637-7segment part is broken. Yesterday it was still okay; today none of the projects or examples work anymore with any library on none of the platforms.

Am I the only one? Is it my slow internet connection? Any clue about what happened?

**Describe the bug**
The 4 digit 7-segment TM1637 part seems not responsive. The simulator start with two segments of digit 0 lit. After init all segments light up, but nothing else happens.

**To Reproduce**
Run any example on any platform with any library.

**Environment (please complete the following information):**
 - OS: tested Android 11/Windows 10
 - Browser: tested using Chrome on tablet, phone and laptop

urish, 2025-03-03:
Thanks for reporting, reproduced with https://wokwi.com/projects/356661328560439297

investigating.

urish, 2025-03-03:
Fixed!

closed 2025-03-03 by urish

maxint-rd, 2025-03-03:
Wow, that was fast. Excellent! (now I can continue testing my TM16xx library to make sure it is fully compatible with Wokwi and also link to Wokwi as a demo platform.

One question though: is the source code of the TM1637 part published somewhere? It would be nice if there were other TM16xx parts in Wokwi, such as the TM1638 LED&KEY module...

urish, 2025-03-03:
Thanks!

It's not - but it's possible to make TM1638 using the [Wokwi Custom Chips API](https://docs.wokwi.com/chips-api/getting-started).

Deepseek is actually pretty good at that, if you give it a relevant example (e.g. of an I2C sensor chip if you are after some I2C chip), a copy of the Chips API documentation, and an example of an Arduino library that uses the target part.

maxint-rd, 2025-03-03:
Okay. Thanks for the suggestion.

And of course thank you for sharing all your amazing work for Wokwi!

I think I saw one of your videos (about simulating an MCU) a couple of years ago and was really impressed by that accomplishment. Although I probably have to view it several times more to get good understanding, it gave me a good idea of the possibilities.
Just now I visited your [YouTube channel](https://www.youtube.com/@Wokwi) and saw that I missed many more interesting video's because I forgot to subscribe back then. Fixed that immediately. 

Thank you so much!