wokwi/wokwi-features #968: Diagram: Click on a part to move, moves all parts when dragging with mouse

Nags-AU, 2025-03-04:
**Describe the bug**
Tried to move a part by clicking on it and then dragging it with the mouse. All parts moved.
(Note: Shift Key or were Select all not used)
Clicking on a part and using keyboard arrows moved the selected part only.
Images Before move, after move with mouse and move with arrows
[image]
[image]
[image]

**Expected behavior**

**Environment (please complete the following information):**
 - OS: Windows 10 
 - Browser Microsoft Edge
 - Version 133.0.3065.92 (Official build) (64-bit)

**Additional context**
Add any other context about the problem here.

urish, 2025-03-04:
Thanks for reporting, investigating.

urish, 2025-03-04:
Issue reproduced and fixed!

closed 2025-03-04 by urish