wokwi/wokwi-features #981: Wokwi-membrame-kepad display of alternative keyboard not working.

NoTreSign, 2025-03-19:
**Describe the bug**
Wokwi-membrame-kepad display of alternative keyboard not working. extracting data from string results in an incorrect display.
A clear and concise description of what the bug is. 

**To Reproduce**
use  : 
```
 {
      "type": "wokwi-membrane-keypad",
      "id": "keypad",
      "top": -597.2,
      "left": -301.6,
      
      "attrs": {
        "keys": ["1","2","3","4","Q","W","E","R","A","S","D","F","Z","X","C","V"]
        
      }
    }
```

dispaly will show:
[    "    1    "
.    "    2    "
.    "    3    "
.    "    4    "

as you can see the string extraction already starts at the [ and will con separate each character separately.
Steps to reproduce the behavior, including link to a simulation project where we can see the problem.

changing the example on https://wokwi.com/projects/344891391763022419  directly shows the issue

**Expected behavior**
a dispaly where the following is shown:

"1","2","3","4",
"Q","W","E","R",
"A","S","D","F",
"Z","X","C","V"

**Environment (please complete the following information):**
 - OS: Windows
 - Browser chrome, edge
 - Version current

**Additional context**
problem was not there 4 months ago

NoTreSign, 2025-03-19:
additionally i found that this also influence the input given to the Arduino code. by removing the key attribute the Arduino code works as intended with the simulation. however when using the key attribute i will get the first row correct. but row 2 , 3 and 4 will be a scramble of the given attribute.

urish, 2025-03-19:
Thanks for the detailed report, fixed!

closed 2025-03-19 by urish