wokwi/wokwi-features #983: Generate a .mov or .gif from wokwi-cli test

rafaelcorsi, 2025-03-26:
Hi, 

It would be nice if we were able to generate a video animation from a given test scenario, similar to the --screenshot-time option.

This would open new possibility to use wokwi-cli with students:

- better debug test
- integration in other platforms as prairielearn

all the best,

urish, 2025-03-27:
Hi Rafael!

You mean generate an animation of the content of an LCD screen, or of the entire diagram?

What's prairielearn?

rafaelcorsi, 2025-03-27:
[prairielearn](https://www.prairielearn.com/)  is a platform that we use in some disciplines in which each exercise is generated by a code, so we can have "infinite" variations of the same exercise. This is used in the context of mastery learning, where the student can only move on to the next stage when he has mastered the current one.

And I have been thinking about how to use wokwi with this resource. Imagine the exercise in which the student needs to blink an LED, for each time he tries to solve the exercise the system generates a new one, in which I vary in the diagram.json the pin to which the LED is connected.

So every time the student tries to solve the exercise the LED will change pin, and he will have to check which GPIO again and change the code. This makes it easier to fix basic concepts that cause problems later on.

In the case of the LED, it's fine to generate a pass/fail result or provide the wokwi-cli log, but for more complex cases, this result alone won't be enough, and so I thought about providing the simulation animation to the students, as a test result.