wokwi/wokwi-features #986: Project gets stuck on "Loading Project.." on VS Code

mustafaakagunduz, 2025-03-27:
Hello everyone,

When running a Wokwi project in VSCode, the simulation(when I launch diagram.json) gets stuck on "Loading Project.." and does not proceed. No errors are displayed, and the project does not start.  

Steps to reproduce the behavior:  
1. Open VSCode and install the Wokwi extension.  
2. Load a Wokwi project (e.g., an ESP32 simulation).  
3. Run the project using the Wokwi extension.  
4. The screen remains stuck on "Loading Project.." indefinitely.  

Expected behavior:
The simulation should start normally, displaying the project interface and running the embedded code.  

Environment (please complete the following information):**  
- OS: macOS  
- VSCode Version: latest
- Wokwi Extension Version**: latest

Additional context
- I have tried reinstalling the Wokwi extension, but the issue persists. 
- Internet connection is stable, and no VPN is enabled.  
- The same project works fine in the web version of Wokwi.  

Any insights or troubleshooting steps would be appreciated! 🚀  

[image]

urish, 2025-03-27:
Hi, thanks for reporting!

Is it just this specific project, or any projects? 

Did it work previously?

Also, can you please try with the following test project, which is known to work: https://github.com/wokwi/esp-idf-hello-world

mustafaakagunduz, 2025-03-27:
Hi, thanks for considering my issue,

I've just tried your diagram in VS Code and it works properly.

I have a Wokwi project, it has 2 files, one of them is "diagram.json" and other is "project.ino" file. When I copy the content of the project.ino to the main.cpp and change the main.cpp file to main.ino, it started work properly..

I didn't understand why but it works 😅 Do you have any comment about it?