wokwi/wokwi-features #988: ESP32-S3 throughs `IllegalInstruction`-error for `dsps_dotprod_f32` due to `dsps_dotprod_f32_ae32`.

GiorgosXou, 2025-03-29:
**Describe the bug**
Based on [this macro-condition](https://github.com/espressif/esp-dsp/blob/74e23cd649c40d7835c9de3fc02f69bc838508e9/modules/dotprod/include/dsps_dotprod.h#L103-L112) `dsps_dotprod_f32` defaults to [`dsps_dotprod_f32_ae32`](https://github.com/espressif/esp-dsp/blob/74e23cd649c40d7835c9de3fc02f69bc838508e9/modules/dotprod/float/dsps_dotprod_f32_ae32.S) for ESP32-S3. However, *(despite the fact that it compiles)* it throughs `IllegalInstruction`-error in log, therefore it doesn't work. Unfortunately, I don't have an ESP32-S3 to verify this 100%, but as far as I understand this should be a wokwi issue. 

**To Reproduce**
Just test [this wokwi example here](https://wokwi.com/projects/426757785193378817):
```cpp
#include "esp_dsp.h"

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Hello, ESP32-S3!");
  
  float x[]  = {1,2,3,4};
  float y[]  = {5,4,3,2};
  float dest = 1;

  dsps_dotprod_f32(x, y, &dest, 4); // FAILS (ASM)
  // dsps_dotprod_f32_ae32(x, y, &dest, 4); // FAILS (ASM)
  // dsps_dotprod_f32_ansi(x, y, &dest, 4); // WORKS (C-CODE)
  
  Serial.println(dest);
}

void loop() {}
```

**Expected behavior**
Based on [this example](https://github.com/espressif/esp-dsp/blob/fe057473b724f788611f11e3d5ce3d412a866998/examples/dotprod/main/dsps_dotproduct_main.c#L54), `dsps_dotprod_f32` *(I assume it)* should work just like the `...ansi`-version works without any error.

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Browser: Ungoogled-chromium
 - Version: -

**Additional context**
As said, unfortunately: I don't have an ESP32-S3 to verify this 100% *(an order I had never came)*, but as far as I understand this should be a wokwi issue.

urish, 2025-03-29:
Thanks for reporting! 

I looked into this, and it seems that the DSP code uses one of the ESP32-S3 PIE (Processor Instruction Extensions), ee.ldf.128.ip, which is not currently implemented in Wokwi. 

Out of curiosity, what project are you working on?

GiorgosXou, 2025-03-29:
> Out of curiosity, what project are you working on?

Thanks for asking. I'm working on my [NeuralNetworks](https://github.com/GiorgosXou/NeuralNetworks)-library and while I was messing around with DSPs and SIMD-instructions for adding extensive support for other CPU-architectures<b><sup>^1</sup></b> + while I was developing RNN support<b><sup>^2</sup></b>, I went back to wokwi to test if everything works fine *(since my order never came)*. And that's how I found this issue, in a nutshell (: 

<b><sup>[1]:</sup></b><sup> *(eg. see issue for [arm_dot_prod_f32](https://stackoverflow.com/questions/79542139/does-pico-2-utilizes-simd-instructions-or-just-loop-unrolling-during-arm-dot-pr))*</sup>
<b><sup>[2]:</sup></b><sup> *(for a private-company [I'm recently collaborating with](https://github.com/GiorgosXou/NeuralNetworks/discussions/48) [not officially yet])*</sup>

urish, 2025-03-29:
Thanks for explaining the use case! 

Implementing the SIMD + DSP instruction simulation would take a considerable amount of effort. Is this something that would be helpful for your work?

GiorgosXou, 2025-03-29:
it'd be cool, but there's no such rush at all, you don't need to. I re-ordered an ESP, so in a few days I'll be able to test the custom function I was planning to make based on those simd instructions. Stay safe ✌

urish, 2025-06-23:
Fixed (we implemented the required instructions on the S3)