wokwi/wokwi-features #992: ESP32: Standard ARDUINO example for RMT  RMT_LED_BLINK ?

fanfanlatulipe26, 2025-04-14:
I try to run the standard Arduino example for RMT  ESP32 RMT_LED_BLINK, as delivered with the Arduino ESP32 3.2.0 and see no blink at all.
I also have a look with the logic analyzer on pin 2 (the one used in the example) and see very little activity.

Don't know on which Espressif core / arduino ESP32 core the similator is built.

WIndows . Mozilla

urish, 2025-04-22:
Hi, thanks for reporting.

Can you please provide a link to to the relevant Wokwi projects so we can try to reproduce it?

fanfanlatulipe26, 2025-04-22:
Here it is [fs_esp32_rmt_blink](https://wokwi.com/projects/428207943957650433)
It is the example comming with the standard Arduino ESP32 distribution: in Arduino IDE 
Files/Examples/ESP32/RMT/RMT_LED_Blink

urish, 2025-04-22:
Thanks. Issue reproduced, working on a fix.

urish, 2025-04-22:
Found one issue and uploaded a fix. It's working almost as expected now, some of the timings are still off, but at least the number of blinks is correct as well as most of the periods.

fanfanlatulipe26, 2025-04-22:
Super. I discover the Wokwi simulator last week when I forgot my ESPs bords during some holidays and wanted to work a bit. 
Nice tool. Congrat. Well done.

urish, 2025-04-23:
Thanks!

Actually, I just checked the example on the physical hardware, and it seems like the issue is in the example code itself. The simulation now behaves just like the actual hardware.

The issue in the example is the usage of `RMT_SYMBOLS_OF(blink_500ms_rmt_data) - 2` instead of `RMT_SYMBOLS_OF(blink_500ms_rmt_data)`, thus skipping the last two RMT signals.

closed 2025-04-23 by urish

fanfanlatulipe26, 2025-04-23:
Running the example on a physical ESP32 I get now a crash after some blinks ... Strange. I filled a new issue for it.