wokwi/wokwi-features #995: I tried to compile your program and got this answer :Multiple libraries were found for "LiquidCrystal_I2C.h"

aledaniel1, 2025-04-20:
**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior, including link to a simulation project where we can see the problem.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: [e.g. iOS/Android/Windows]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

aledaniel1, 2025-04-20:
I tried to compile your program and got this answer :Multiple libraries were found for "LiquidCrystal_I2C.h"

urish, 2025-04-21:
Link to the project?