wokwi/wokwi-features #999: hardware/rtc.h not found on pico builds

alien999999999, 2025-05-01:
**Describe the bug**
when you choose a pico build and need RTC support, it doesn't work; compile says it cannot find hardware/rtc.h

**To Reproduce**
start a new pico build and add hardware/rtc.h in the includes

**Expected behavior**
hardware/rtc.h to be found

**Environment (please complete the following information):**
 - OS: NA
 - Browser: Firefox
 - Version: ESR 128.7

**Additional context**
Documentation states that RTC is support with pico build

urish, 2025-05-05:
Hi, can you please provide a link to a wokwi project showing the problem?

Also, have are you using Pi Pico SDK project template? https://wokwi.com/projects/new/pi-pico-sdk

I just checked and was able to include "hardware/rtc.h" there

alien999999999, 2025-05-05:
I'm pretty sure I had started from the pi-pico-sdk link, but for some reason, i can't find that anymore starting from the https://wokwi.com page... I'm trying to replicate it from the start

alien999999999, 2025-05-05:
this compile error (not from scratch) shows the error:
```
PICO_SDK_PATH is /home/wokwi/pico-sdk
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/wokwi/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Build type is Release
TinyUSB available at /home/wokwi/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/wokwi/pico-sdk/lib/btstack
cyw43-driver available at /home/wokwi/pico-sdk/lib/cyw43-driver
lwIP available at /home/wokwi/pico-sdk/lib/lwip
mbedtls available at /home/wokwi/pico-sdk/lib/mbedtls
CMake Warning at /home/wokwi/pico-sdk/tools/Findpicotool.cmake:30 (message):
  No installed picotool with version 2.1.1 found - building from source

  It is recommended to build and install picotool separately, or to set
  PICOTOOL_FETCH_FROM_GIT_PATH to a common directory for all your SDK
  projects
Call Stack (most recent call first):
  /home/wokwi/pico-sdk/tools/CMakeLists.txt:138 (find_package)
  /home/wokwi/pico-sdk/tools/CMakeLists.txt:493 (pico_init_picotool)
  /home/wokwi/pico-sdk/src/cmake/on_device.cmake:56 (picotool_postprocess_binary)
  CMakeLists.txt:12 (pico_add_extra_outputs)

Downloading Picotool
/home/wokwi/project/src/main.c:3:10: fatal error: hardware/rtc.h: No such file or directory
    3 | #include "hardware/rtc.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/wokwi_project.dir/build.make:79: CMakeFiles/wokwi_project.dir/src/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2259: CMakeFiles/wokwi_project.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

Error: Process exited with 2
```

this is the beginning of the main.c file:

```
#include "pico/stdlib.h"
#include "pico/time.h"
#include "hardware/rtc.h"
#include "hardware/gpio.h"
//#include "hardware/pio.h"
#include "hardware/irq.h"
#include "hardware/spi.h"
#include "pico/binary_info.h"
```

urish, 2025-05-05:
Thanks! I added the hardware_rtc library to the pico SDK builder.

Note that for existing projects, you may have to slightly change the code (e.g. add some comment) to trigger a rebuild.

closed 2025-05-05 by urish

alien999999999, 2025-05-05:
hey, would it not also be better to have a template link to https://wokwi.com/projects/new/pi-pico-sdk as well? or is it hidden on purpose ( from the https://wokwi.com/pi-pico page ), also no C language button from the languages part on the main website?

urish, 2025-05-05:
Added to /pi-pico.

Most people use Wokwi online with the Arduino Wiring-based Framework, which is C++. From my experience, including a C language button will cause more confusion than good.