wokwi/wokwi-features #1000: NVS & Sysloop not working on ESP32 C6 with Rust

Julien-cpsn, 2025-05-04:
**Describe the bug**
When adding either the `sysloop` or the `nvs_default_partition` line, I get the following stack trace. Please note that on the trace, "Hello World!" is not event logged. When commenting `sysloop` or `nvs_default_partition`, everything works fine. I get the same problem with a new blank project or with the project I am working on.

main.rs
```rust
use esp_idf_svc::wifi::{AuthMethod, ClientConfiguration, Configuration};
use esp_idf_svc::hal::peripherals::Peripherals;
use esp_idf_svc::eventloop::EspSystemEventLoop;
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::wifi::EspWifi;
use esp_idf_svc::sys::EspError;

fn main() -> Result<(), EspError> {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    log::info!("Hello World!");
    
    let peripherals = Peripherals::take()?;
    let sysloop = EspSystemEventLoop::take()?;
    let nvs_default_partition = EspDefaultNvsPartition::take()?;

    let mut wifi = EspWifi::new(
        peripherals.modem,
        sysloop.clone(),
        Some(nvs_default_partition.clone()),
    )?;

    wifi.set_configuration(&Configuration::Client(ClientConfiguration {
        ssid: String::from("Wokwi-GUEST").parse().unwrap(),
        password: String::from("").parse().unwrap(),
        auth_method: AuthMethod::None,
        ..Default::default()
    }))?;

    wifi.start()?;
    wifi.connect()?;

    Ok(())
}
```

PS: Example WiFi code taken from Wokwi documentation but updated to fit more recent crates usage

**To Reproduce**
Initialize a new ESP32-C6 project, copy and paste the main.rs file I provided.

**Expected behavior**
The program should at least load, run and output "Hello World!"

**Environment (please complete the following information):**
 - OS: Debian + Nix
 - Jetbrains RustRover extension
 - Versions
   - Wokwi extension v0.10.4
   - ESP-IDF v5.4.1

**Additional context**

Here is the trace:
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x5c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x126c
load:0x4086c410,len:0x704
load:0x4086e610,len:0x2ae0
entry 0x4086c410
I (49) cpu_start: Unicore app
E (25) cpu_start: Invalid app image header

abort() was called at PC 0x4080153d on core 0
Core  0 register dump:
MEPC    : 0x40801a10  RA      : 0x408074ee  SP      : 0x4087e160  GP      : 0x4080dcc4  
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x4087e19c  S1      : 0x4087e19c  A0      : 0x4087e19c  A1      : 0x4087e17e  
A2      : 0x00000000  A3      : 0x4087e1c9  A4      : 0x00000001  A5      : 0x40811000  
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4087e180  S3      : 0x0000004d  
S4      : 0x42078000  S5      : 0x42000000  S6      : 0x408013a4  S7      : 0x00072fa8  
S8      : 0x0004f554  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00000080  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
4087e160: 0x00000000 0x00000000 0x4087e19c 0x4080c38e 0x408013a4 0x42000000 0x42078000 0x00000030
4087e180: 0x30383034 0x64333531 0x00300000 0x4080f424 0x4087e180 0x4080f440 0x4087e17c 0x726f6261
4087e1a0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x31303830 0x20643335
4087e1c0: 0x63206e6f 0x2065726f 0x00000030 0x00000000 0x42800000 0x00000001 0x00300000 0x40801540
4087e1e0: 0x00000000 0x4004b05c 0x0000ffff 0x00700000 0x42800000 0x420c7573 0x42072fc7 0x42000000
4087e200: 0x42800000 0x420c7573 0x42072fc7 0x4086f88c 0x00000000 0x00078000 0x00018020 0x00070020
4087e220: 0x42078020 0x42000020 0x0004f554 0x00072fa8 0x00004480 0x4004b05c 0x00000000 0x40880000
4087e240: 0x00004480 0x4004b05c 0x0000ffff 0xffffffff 0xffffffff 0x4087e3a0 0xffffffff 0x4086fbba
4087e260: 0x00000000 0x00000000 0x00010000 0x00190000 0x00010000 0x000006e9 0x408013a4 0x00000000
4087e280: 0x0000000d 0x00000000 0x00000000 0x40800000 0x00007ff8 0x42078020 0x0004f554 0x40807ff8
4087e2a0: 0x000054c0 0x4080d4c0 0x00002120 0x00000000 0x000014ac 0x42000020 0x00072fa8 0x00000000
4087e2c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e2e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e300: 0x00000000 0x00000000 0x00000000 0x00010020 0x00018020 0x0006757c 0x0006ca44 0x0006eb6c
4087e320: 0x00070020 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e340: 0x00000000 0x00000000 0x00000000 0x000d2fd0 0x00000000 0x00000000 0x00000000 0x00000000
4087e360: 0x00000000 0x00000000 0x00000000 0x00000000 0xffffffff 0x00000010 0x0000000a 0x40880000
4087e380: 0x00000000 0x000000d1 0x00004490 0x4086c492 0x00000000 0x00000000 0x00000000 0xffffffff
4087e3a0: 0x00000000 0x00000000 0x00010000 0x00190000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e440: 0x60002000 0x00000100 0x00000020 0x4001b1c0 0x00000000 0x00000000 0x00000000 0x00000000
4087e460: 0x00000000 0x00000000 0x00010000 0x400203e9 0x4086c410 0x4086e610 0x00002ae0 0x000000ee
4087e480: 0x0000000d 0x00006300 0x01000000 0x42004490 0x7533885e 0x7a878fdd 0x2652fa7c 0xdd2f3bd0
4087e4a0: 0x156c3f66 0x8e4aedb2 0x224ab713 0xc1f9d587 0x4c87ccfb 0x00000000 0x00000000 0x00000000
4087e4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4087e5f0 0x00000101 0x00000002
4087e4e0: 0x7a878fdd 0x2652fa7c 0xdd2f3bd0 0x156c3f66 0x8e4aedb2 0x224ab713 0xc1f9d587 0x4c87ccfb
4087e500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e520: 0x00000000 0x00000000 0x00000000 0xd1000000 0x00000080 0x00000000 0x00000000 0x00000000
```

My Cargo.toml file:
```toml
[package]
name = "test"
version = "0.1.0"
authors = ["Julien-cpsn <julien.caposiena@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.77"

[[bin]]
name = "test"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = []

experimental = ["esp-idf-svc/experimental"]

[dependencies]
log = "0.4"
esp-idf-svc = { version = "0.51", features = ["critical-section", "embassy-time-driver", "embassy-sync"] }

[build-dependencies]
embuild = "0.33"
```

diagram.json
```json
{
    "version": 1,
    "editor": "wokwi",
    "author": "Julien-cpsn <julien.caposiena@gmail.com>",
    "parts": [
        {
            "type": "board-esp32-c6-devkitc-1",
            "id": "esp",
            "top": 0.59,
            "left": 0.67,
            "attrs": { "flashSize": "16" }
        }
    ],
    "connections": [
        ["esp:TX", "$serialMonitor:RX", "", []],
        ["esp:RX", "$serialMonitor:TX", "", []]
    ],
    "serialMonitor": {
        "display": "terminal",
        "convertEol": true
    }
}
```

urish, 2025-05-05:
Thanks for reporting!

Can you please also share the compiled firmware elf file to aid in debugging this?

Julien-cpsn, 2025-05-05:
Here is the link to the ELF file: https://we.tl/t-sxcn1y3GH1

```
file target/riscv32imac-esp-espidf/debug/test
target/riscv32imac-esp-espidf/debug/test: ELF 32-bit LSB executable, UCB RISC-V, RVC, soft-float ABI, version 1 (SYSV), statically linked, with debug_info, not stripped
```

urish, 2025-05-05:
Interesting, it seems to panic at `uart_ll_set_baudrate`: 

Thread 1 hit Breakpoint 1, panic_abort (
    details=details@entry=0x4087e19c "abort() was called at PC 0x4080159d on core 0")
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/esp_system/panic.c:468
warning: 468    /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/esp_system/panic.c: No such file or directory
(gdb) bt
#0  panic_abort (
    details=details@entry=0x4087e19c "abort() was called at PC 0x4080159d on core 0")
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/esp_system/panic.c:468
#1  0x40806c4a in esp_system_abort (
    details=details@entry=0x4087e19c "abort() was called at PC 0x4080159d on core 0")
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/esp_system/port/esp_system_chip.c:87
#2  0x4080d168 in abort ()
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/newlib/abort.c:38
#3  0x408015a0 in uart_ll_set_baudrate (hw=<optimized out>,
    baud=<optimized out>, sclk_freq=<optimized out>)
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/hal/esp32c6/include/hal/uart_ll.h:409
#4  call_start_cpu0 ()
    at /nix/store/xi861kdg5z7x9bxzza7s0ccxl0pgvk0m-esp-idf-v5.4.1/components/esp_system/port/cpu_start.c:743

urish, 2025-05-05:
I've just flashed your .elf file to a physical ESP32-C6 board, and it results in the same exception:

```
$ espflash monitor 
[2025-05-05T09:10:25Z INFO ] Serial port: 'COM3'
[2025-05-05T09:10:25Z INFO ] Connecting...
[2025-05-05T09:10:33Z INFO ] Using flash stub
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d68
load:0x40875720,len:0x1800
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader     
I (24) boot: compile time Jun  7 2023 08:02:08
I (25) boot: chip revision: v0.0
I (29) boot.esp32c6: SPI Speed      : 40MHz
I (33) boot.esp32c6: SPI Mode       : DIO
I (38) boot.esp32c6: SPI Flash Size : 4MB
I (43) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000     
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 003f0000     
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=c5d88h (810376) map
I (259) esp_image: segment 1: paddr=000d5db0 vaddr=40800000 size=02268h (  8808) load
I (262) esp_image: segment 2: paddr=000d8020 vaddr=420c8020 size=3d898h (252056) map
I (316) esp_image: segment 3: paddr=001158c0 vaddr=40802268 size=15614h ( 87572) load
I (336) esp_image: segment 4: paddr=0012aedc vaddr=40817880 size=04188h ( 16776) load
I (346) boot: Loaded app from partition at offset 0x10000
I (347) boot: Disabling RNG early entropy source...
I (358) cpu_start: Unicore app
E (367) cpu_start: Invalid app image header

abort() was called at PC 0x4080159d on core 0
Core  0 register dump:
MEPC    : 0x40801a70  RA      : 0x40806c4a  SP      : 0x4087e160  GP      : 0x40818084
TP      : 0x00000000  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x4087e19c  S1      : 0x4087e19c  A0      : 0x4087e19c  A1      : 0x4087e17e
A2      : 0x00000000  A3      : 0x4087e1c9  A4      : 0x00000001  A5      : 0x40821000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4087e180  S3      : 0x0000004d  
S4      : 0x42000000  S5      : 0x42000000  S6      : 0x40801404  S7      : 0x0003d898
S8      : 0x000c5d88  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
4087e160: 0x00000000 0x00000000 0x4087e19c 0x4080d168 0x40801404 0x42000000 0x42000000 0x00000030
4087e180: 0x30383034 0x64393531 0x00300000 0x4081b540 0x4087e180 0x4081b55c 0x4087e17c 0x726f6261
4087e1a0: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x31303830 0x20643935
4087e1c0: 0x63206e6f 0x2065726f 0x00000030 0x00000000 0x42800000 0x00000001 0x00300000 0x408015a0
4087e1e0: 0x000052e0 0x4004b05c 0x0000ffff 0x00700000 0x42800000 0x420c5da7 0x421058b7 0x420c8000
4087e200: 0x42800000 0x420c5da7 0x421058b7 0x4086f95c 0x13d07089 0x00040000 0x00010020 0x000d8020
4087e220: 0x42000020 0x420c8020 0x000c5d88 0x0003d898 0x000052e0 0x0000015a 0x00000000 0x40880000
4087e240: 0x000052e0 0x4004b05c 0x0000ffff 0xffffffff 0xffffffff 0x4087e3a0 0xffffffff 0x4086fda2
4087e260: 0xffffffff 0x0000002e 0x00010000 0x003f0000 0x00010000 0x200205e9 0x40801404 0x000000ee
4087e280: 0x0000000d 0x00006300 0x01000000 0x42000020 0x000c5d88 0x40800000 0x00002268 0x420c8020
4087e2a0: 0x0003d898 0x40802268 0x00015614 0x40817880 0x00004188 0x00000000 0x00000000 0x00000000
4087e2c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e2e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e300: 0x00000000 0x00000000 0x00000000 0x00010020 0x000d5db0 0x000d8020 0x001158c0 0x0012aedc
4087e320: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e340: 0x00000000 0x00000000 0x00000000 0x0011f090 0x9f9741f4 0xa26b2fa7 0x403c9039 0xc75bd871
4087e360: 0x2b320c24 0xa9faeda0 0xf789fd4c 0xca4e0a5e 0x000052e0 0x4004b05c 0x0000ffff 0x40880000
4087e380: 0x00000000 0x00000021 0x000052f0 0x4086c492 0xc4480606 0x6f38ea77 0x00000000 0xffffffff
4087e3a0: 0x00000000 0x00000000 0x00010000 0x003f0000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e440: 0x60002000 0x00000100 0x00000020 0x4001b1c0 0x0dbd5424 0xcd897671 0x3dae3f3f 0x00000000
4087e460: 0xf354ba51 0x00000000 0x00010000 0x200203e9 0x4086c410 0x40875720 0x00001800 0x000000ee
4087e480: 0x0000000d 0x00006300 0x01000000 0x420052f0 0x7533885e 0x7898f8af 0xf5bf6ca9 0x8ac3667c
4087e4a0: 0x22a4d1a6 0x405078b7 0x58eaefb5 0x6cefab2b 0xc2277435 0x00000000 0x00000000 0x00000000
4087e4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4087e5f0 0x00000101 0x00000002
4087e4e0: 0x7898f8af 0xf5bf6ca9 0x8ac3667c 0x22a4d1a6 0x405078b7 0x58eaefb5 0x6cefab2b 0xc2277435
4087e500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4087e520: 0x00000063 0x5f707365 0x5f6d6f72 0x69676572 0x775f6332 0x65746972 0x73616d5f 0x0000006b
```

So it doesn't look like an issue with Wokwi.

closed 2025-05-05 by urish