wokwi/wokwi-features #1001: I2C bus for STM32 does no longer work.

Koepel, 2025-05-07:
The normal I2C bus or the created TwoWire bus do no longer work with a I2C display on the STM32 Nucleo64 C031C6 board.

For example the second example in the 'Docs' does no longer work: https://docs.wokwi.com/parts/board-st-nucleo-c031c6
I prefer the normal I2C bus with the normal library. This is my test project: https://wokwi.com/projects/430302099906345985

My test project starts alright, but the SCL is kept low forever after the first data.

[image]

Tested only with: STM32 Nucleo64 C031C6 and only with a I2C LCD display.

ScoutLED, 2025-12-01:
Is there any update for this issue?

ScoutLED, 2025-12-01:
Well, thank you for your clarification.
Hopefully, it is gonna be fixed in the future.