wokwi/wokwi-features #1002: Question about WOKWI network

joshuasi546, 2025-05-08:
is it possible if I want the simulated ESP32 device to connect to the same network as my computer for device discovery? I'm aware that it uses a virtual WiFi access point called Wokwi-GUEST, is there a way so that the simulated device uses my computer network instead?

urish, 2025-05-08:
There is an experimental version of the Private IoT Gateway that can support this, but it only works on Linux in certain configurations. Which operating system do you have?

joshuasi546, 2025-05-09:
Currently trying on Ubuntu 24.04.1 LTS

joshuasi546, 2025-05-15:
I assume that the current stable Wokwi feature does not support this functionality? you can close the issue if it's correct

joshuasi546, 2025-05-26:
Got it, we will let you know if we decided to go into that route. Thank you again for the assist really appreciate your help.

closed 2025-05-26 by joshuasi546