wokwi/wokwi-features #1004: Diferent behavor between website and vscode extension (rp2040 - timer)

rafaelcorsi, 2025-05-12:
Target: Rasp Pico 1 sdk
Example project: https://wokwi.com/projects/430756341381772289

Code:

When I run it on the webpage I got the timers been canceled and created as expected, but when I run on the vs code extension, it won't create a new timer. To solve the problem I need to add a sleep between cancel and add a new timer.

```diff
        cancel_repeating_timer(&timer_b);
+       sleep_ms(10);
        add_repeating_timer_ms(300, timer_b_callback, NULL, &timer_b);
                 sleep_ms(1000);
```

Example code:

```c
#include <stdio.h>
#include "pico/stdlib.h"

const int LED_PIN_R = 4;

volatile int cnt = 0;

bool timer_b_callback(repeating_timer_t* rt) {
    printf("timer %d\n", cnt++);
    return true; // keep repeating
}

int main() {
    stdio_init_all();
    gpio_init(LED_PIN_R);
    gpio_set_dir(LED_PIN_R, GPIO_OUT);

    repeating_timer_t timer_b;

    add_repeating_timer_ms(100, timer_b_callback, NULL, &timer_b);
    sleep_ms(1000);

    while (true) {
      printf("Canceled timer\n");
        cancel_repeating_timer(&timer_b);
        add_repeating_timer_ms(300, timer_b_callback, NULL, &timer_b);
                 sleep_ms(1000);

    }
}
```

online it creates the new timer:

[image]

on vscode it get stuck

[image]

rafaelcorsi, 2025-05-27:
Hello @urish sorry for the delay, it works fine on the SDK 2.1.1

Is there something that I can do to fix this? Students use the first version they see on the pico extension, so I can't control this...

urish, 2025-06-02:
Which SDK version exactly does this reproduce with?

I need to be able to reproduce locally in order to be able to investigate

closed 2025-10-26 by urish