wokwi/wokwi-features #1005: ESP_FAIL when using esp-idf-hal (rust) I2C

jacobaclarke, 2025-05-24:
**Describe the bug**
I2C fails when using esp-idf-hal (rust) and board-esp32-devkit-c-v4. I have tested this code on a physical esp32 without error.  

**To Reproduce**
```{json}
{
  "version": 1,
  "author": "Minimal Example",
  "editor": "wokwi",
  "parts": [
    {
      "type": "board-esp32-devkit-c-v4",
      "id": "esp",
      "top": 0,
      "left": 0,
      "attrs": { "flashSize": "16" }
    }
  ],
  "connections": [
    ["esp:TX", "$serialMonitor:RX", "", []],
    ["esp:RX", "$serialMonitor:TX", "", []],
    ["esp:21", "esp:18", "green", ["h-40", "v-40"]],
    ["esp:22", "esp:19", "yellow", ["h40", "v40"]]
  ],
  "serialMonitor": { "display": "terminal", "convertEol": true },
  "dependencies": {}
}
```

```{rust}
use esp_idf_svc::hal::prelude::*;
use anyhow::Result;
use esp_idf_hal::{
    delay::BLOCK,
    i2c::{I2cConfig, I2cDriver, I2cSlaveConfig, I2cSlaveDriver},
    gpio::AnyIOPin,
};

fn main() -> Result<()> {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take().unwrap();

    const SLAVE_ADDR: u8 = 0x22;
    const SLAVE_BUFFER_SIZE: usize = 256;

    // Explicitly convert pins to AnyIOPin
    let sda_slave: AnyIOPin = peripherals.pins.gpio18.into();
    let scl_slave: AnyIOPin = peripherals.pins.gpio19.into();
    let sda_master: AnyIOPin = peripherals.pins.gpio21.into();
    let scl_master: AnyIOPin = peripherals.pins.gpio22.into();

    // Initialize I2C slave (I2C1)
    let slave_config = I2cSlaveConfig::new()
        .rx_buffer_length(SLAVE_BUFFER_SIZE)
        .tx_buffer_length(SLAVE_BUFFER_SIZE);
    let mut i2c_slave = I2cSlaveDriver::new(
        peripherals.i2c1,
        sda_slave,
        scl_slave,
        SLAVE_ADDR,
        &slave_config,
    )?;

    // Initialize I2C master (I2C0)
    let master_config = I2cConfig::new().baudrate(100.kHz().into());
    let mut i2c_master = I2cDriver::new(
        peripherals.i2c0,
        sda_master,
        scl_master,
        &master_config,
    )?;

    // Test data
    let tx_buf: [u8; 8] = [0x01, 0x23, 0x45, 0x67, 0x89, 0xab, 0xcd, 0xef];

    // Master write test
    println!("Master write test");
    i2c_master.write(SLAVE_ADDR, &tx_buf, BLOCK)?;
    let mut rx_buf: [u8; 8] = [0; 8];
    println!("Slave read test");
    i2c_slave.read(&mut rx_buf, BLOCK)?;
    println!("Master write test: {:?} -> {:?}", tx_buf, rx_buf);

    // Master read test
    println!("Slave write test");
    i2c_slave.write(&tx_buf, BLOCK)?;
    let mut rx_buf: [u8; 8] = [0; 8];
    println!("Master read test");
    i2c_master.read(SLAVE_ADDR, &mut rx_buf, BLOCK)?;
    println!("Master read test: {:?} -> {:?}", tx_buf, rx_buf);

    Ok(())
} 
```

**Expected behavior**
```
Master write test
Slave read test
Master write test: [1, 35, 69, 103, 137, 171, 205, 239] -> [1, 35, 69, 103, 137, 171, 205, 239]
Slave write test
Master read test
Master read test: [1, 35, 69, 103, 137, 171, 205, 239] -> [1, 35, 69, 103, 137, 171, 205, 239]
```

**Environment (please complete the following information):**
 - OS: [e.g. iOS/Android/Windows] Macos
 - Browser [e.g. chrome, safari] VSCode Extension
 - Version [e.g. 22]

**Additional context**
I get an ESP_FAIL error every time I try to run this code.

urish, 2025-05-24:
Thanks for reporting! Mind putting together a repo with the complete project so we can reproduce quickly?

jacobaclarke, 2025-05-24:
Sure! Here it is https://github.com/jacobaclarke/i2c-example

urish, 2025-05-24:
Thank you very much!

closed 2025-05-24 by jacobaclarke

urish, 2025-05-25:
Indeed, thanks for updating. Note that we plan to eventually support I2C slave on the ESP32 family.