wokwi/wokwi-features #1006: A visual presentation mode - Panel mode

Retroperra, 2025-05-26:
WHAT:
It would be great to further show and simulate a design with a panel mode.
The panel mode would hide all non interactive components like connections, microprocessors, resistors etc and keep only displays, LEDs, buttons, pots, encoders etc.

In the very fast and not-worked-on example I switch between the normal mode and a panel mode.

WHY:
As soon as you have multiple buttons, a data bus, and what not the simulated circuit get overwhelming. 
It is not easy to present the idea.

As a UI developer I understand the challenge and at least some of the obstacles  but at least I had to show the idea. 

Here's a link to a video showing what I'm talking about

[Inspelning-2025-05-26-150717.mp4 linked via Dropbox](https://www.dropbox.com/scl/fi/166scxxkrw4k15sa78fmj/Inspelning-2025-05-26-150717.mp4?rlkey=xkchkg2z1s5e6c1a0r59tqzju&st=kkk0tlex&dl=0
)

urish, 2025-06-01:
Thanks for sharing your idea! 

We have something similar on our roadmap: the ability to create custom boards (#302), where you can group some parts (initially LEDs, optionally a microcontroler/custom chips, displays, and sensors, and later also buttons / 7-segment displays and other elements). This is planned to happen sometime in late 2025.

closed 2025-06-01 by urish

urish, 2026-01-04:
Still in the planning phase :-)