wokwi/wokwi-features #1012: Dark mode is no longer sticky

sutaburosu, 2025-06-05:
**Describe the bug**
As of ~3 days ago, each project I open defaults to light mode. This didn't used to happen; my dark/light choice was remembered and opening a project would default to the last choice. docs.wokwi.com still correctly remembers my last choice.

**To Reproduce**
Select dark mode. Close the tab. Open the same link again, and it will be light mode.

**Expected behavior**
The previous behaviour: remember the last chosen dark/light mode, and use it when opening a new page.

**Environment (please complete the following information):**
 - OS: Linux/Wayland/Plasma
 - Browser: Firefox/Chromium

urish, 2025-06-06:
Thanks for reporting! How about now?

sutaburosu, 2025-06-06:
All good now. Thanks, Uri.

closed 2025-06-06 by sutaburosu