wokwi/wokwi-features #1015: ESP-NOW and Multi-Microcontroller Support in Single Project

Peelaphat2544, 2025-06-07:
1. ESP-NOW Protocol Support
- Enable ESP-NOW communication protocol in Wokwi simulator
- Support for peer-to-peer communication between ESP32 devices
- Real-time data transmission simulation without WiFi infrastructure
- Configurable device pairing and communication parameters

2. Multi-Microcontroller Project Support
- Allow multiple microcontroller instances within a single project
- Support for different microcontroller types (ESP32, Arduino, etc.) in one simulation
- Independent code execution for each microcontroller
- Shared simulation environment with inter-device communication

Benefits for Educational and Development Use :
- Simulate IoT mesh networks and distributed systems
- Test multi-device communication protocols
- Prototype complex embedded systems with multiple nodes
- Demonstrate real-world scenarios with device-to-device communication

urish, 2025-06-08:
Thanks for the request, there are several separate issues tracking the requested features:

* #446 
* #186

closed 2025-06-08 by urish

TimiksOMG, 2025-06-12:
So how do you add different files to different boards? I know how to add different boards.
If i add another .ino file with setup and loop, it gives me an error "redefinition of setup (and loop)"
Do i have to use cpp or c files for different controllers? (sorry for bad english btw)

urish, 2025-06-21:
It's not yet supported.