wokwi/wokwi-features #1017: GPIO interrupt and ADC does not work on ESP32 P4

JakubAndrysek, 2025-06-16:
**Describe the bug**
I have written a test that uses GPIO interrupt handling and ADC reading on the ESP32 P4 board. The code works fine on other boards, but it does not work on the P4 board. The GPIO interrupt does not trigger, and the ADC readings are not displayed in the terminal.

**To Reproduce**
There are links to the wokwi projects that demonstrate the issue:
1. [GPIO Interrupt Test](https://wokwi.com/projects/433650769477101569) - (board-esp32-p4-function-ev)
2. [GPIO Interrupt Test 2](https://wokwi.com/projects/433657738659302401) - this version does not show terminal output (board-esp32-p4-previewboard-esp32-p4-preview)
3. [ADC Test with Terminal Output](https://wokwi.com/projects/433656838948487169) 

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Browser: chrome
 - Version: Version 137.0.7151.68 (Official Build) (64-bit)

urish, 2025-06-16:
Thanks for the detailed report!

For GPIO Interrupt Test 2, you can get the serial output to show by changing the first two connections in diagram.json as follows:

```json
    [ "esp32:RX", "$serialMonitor:TX", "", [] ],
    [ "esp32:TX", "$serialMonitor:RX", "", [] ],
```

As for GPIO interrupts / ADC, investigating.

JakubAndrysek, 2025-06-16:
> Thanks for the detailed report!
> 
> For GPIO Interrupt Test 2, you can get the serial output to show by changing the first two connections in diagram.json as follows:
> 
>     [ "esp32:RX", "$serialMonitor:TX", "", [] ],
>     [ "esp32:TX", "$serialMonitor:RX", "", [] ],

Example  GPIO Interrupt Test 2 updated

> As for GPIO interrupts / ADC, investigating.

urish, 2025-06-16:
ADC should now work correctly on the ESP32-P4. Now investigating GPIOs interrupts.

urish, 2025-06-16:
Also fixed the GPIO interrupts on the P4.