wokwi/wokwi-features #1024: Wokwi crashes on Swift code, probably simulating a TFT display

libewa, 2025-07-06:
**Describe the bug**
When running my project, Wokwi consistently crashes setting the D/C pin.

**To Reproduce**
- Download my project: <https://github.com/libewa/swiftflag>
- Install and use a nightly [Swift] toolchain
- Build for ESP32-C3
- Run the project with Wokwi CLI

**Expected behavior**
Wokwi should simulate the ESP32 and TFT waving a flag.

**Environment (please complete the following information):**
 - OS: macOS Sequoia 15.5
 - Browser n/a
 - Version 0.16.2

[Swift]: https://swift.org

urish, 2025-07-06:
Thanks for reporting! 

Does the same code work on a physical device? 
And does it work in Wokwi for VS Code?

Can you please attach a pre compiled binary + elf file?

urish, 2025-07-06:
You can zip them and then attach

libewa, 2025-07-06:
Here you go:
[Archiv.zip](https://github.com/user-attachments/files/21090711/Archiv.zip)

urish, 2025-07-07:
Pushed a fix, now the simulator does not hang anymore when trying to use GPIO11 on the ESP32-C3 DevKit M-1

However, with your code, there's still nothing on the screen. It looks the the SPI in your code is not configured to use the correct pin numbers, and thus, the data does not go to the display:

[image]

urish, 2025-07-09:
Pause the simulation