wokwi/wokwi-features #1030: LDR Module DO Pin Not Responding to Lux Slider in Simulation

gbtucson, 2025-07-18:
I'm using the wokwi-photoresistor-sensor module in a simulation with an ESP32 DevKit. I’ve connected the DO pin to GPIO22, and expected it to output HIGH when bright, LOW when dark, based on the ambient light simulated by the lux slider.
However, I’m seeing that the DO pin remains stuck at a fixed value (usually HIGH), regardless of how I adjust the lux slider. The serial monitor shows no response, and the code using digitalRead() on GPIO22 doesn't detect transitions. The module is properly powered (VCC and GND), and wiring was verified.
What I tried:
- Swapped from DO to AO with a voltage divider (LDR + 10kΩ resistor) wired to GPIO22 — this setup works perfectly.
- Re-tested DO pin with a minimal sketch that prints digitalRead(22) — still no change as lux slider moves.
- Verified that the module is connected to esp32, and the rest of the simulation is functioning (switch input, relay output).
Expected behavior:
DO pin should respond to lux changes based on its internal comparator, and flip states accordingly.
Request:
Could you confirm whether the DO pin is currently simulated to respond to ambient light changes? If there's a fixed internal threshold or missing configuration for the comparator, exposing that or clarifying the behavior would help. Thanks!

urish, 2025-07-19:
Hi, thanks for reporting! 

Can you please include a link to a Wokwi project showing the issue you describe?

Koepel, 2025-07-23:
I don't have such a LDR module, so I don't know if the digital output turns high with light or with darkness.
The digital output changes in the middle of the voltage.

Working example with Arduino Uno: https://wokwi.com/projects/437279678005637121
Working example with ESP32 and pin 22: https://wokwi.com/projects/437279685733643265

gbtucson, can this Issue be closed ? You can join the Discord channel to fix your project.

gbtucson, 2025-07-23:
For testing purposes I changed the relay to an LED and it works perfectly for that, so may be an issue with the relay. Anyway, I have it working in the real world so not an issue for me any longer. Hope my report helps.