wokwi/wokwi-features #1038: ESP-IDF projects failing to link multiple files

adamfk, 2025-08-30:
**Describe the bug**
Esp-idf projects fail to link that have multiple files.

Bug confirmed on discord: https://discord.com/channels/787627282663211009/1109794342513672222/threads/1402834319369834516

**To Reproduce**
Here's a minimal reproduction. I forked the official wokwi ESP-IDF FreeRTOS blinker project and simply moved the blinker code from main.c to blinker.c/h and it fails to link.

https://wokwi.com/projects/438589876278733825

urish, 2025-09-10:
Thanks for reporting, issue reproduced. 

The challenge is that fixing this (by running cmake each time) is going to significantly slow down the build and increase compilation time. I'm looking for a solution that won't have such a dramatic performance impact.

preston-manwaring, 2025-10-01:
@urish, I haven't had to deal with this problem as I don't run a website you you do. So I brainstormed with Claude for a minute. It had some interesting ideas. I assume CMakeLists.txt is being auto-generated by a method running in your site. If this is true, and cmake needs to be called to recreate the makefiles, you could create a checksum script that runs on CMakeLists.txt to look for changes. Then run cmake if the checksum changed. The slowdown you are worried about would only occur when people add new files to the project. Another option is to create a makefile that watches CMakeLists.txt and runs cmake when it changes. Example code for the checksum is below:

```
#!/bin/bash
BUILD_DIR="build"
CMAKE_FILES="CMakeLists.txt cmake/*.cmake"
CHECKSUM_FILE="$BUILD_DIR/.cmake_checksum"

# Calculate current checksum
CURRENT_CHECKSUM=$(find . -name "CMakeLists.txt" -o -name "*.cmake" | \
                   sort | xargs cat | md5sum)

# Check if we need to reconfigure
if [ ! -f "$CHECKSUM_FILE" ] || [ "$CURRENT_CHECKSUM" != "$(cat $CHECKSUM_FILE)" ]; then
    echo "CMake files changed, reconfiguring..."
    cmake -B $BUILD_DIR -S .
    echo "$CURRENT_CHECKSUM" > "$CHECKSUM_FILE"
else
    echo "No CMake changes detected, skipping reconfiguration"
fi

# Build
cmake --build $BUILD_DIR
```

closed 2025-10-23 by urish