wokwi/wokwi-features #1043: Potentiometer does no longer work in Firefox browser.

Koepel, 2025-09-17:
Test project: https://wokwi.com/projects/442381734344109057

After starting the simulation, the round potentiometer can not be changed in the Firefox browser. It stays at the zero position. When clicking on it a few times, it jumps to the middle position and stays there.
Firefox browser version 143.0

It works normal in Chrome based browsers.

Koepel, 2025-09-18:
This Issue was already mentioned: https://github.com/wokwi/wokwi-features/issues/1042
Please merge. Sorry for not checking existing Issues.

**Additional browser test:**
I have tested many browsers in linux, Windows and Android.
**The problem was introduced with Firefox version 143.0** in linux, Windows and Android. That means that it is a bug in Firefox or a compatibility problem.
The previous version 142.0.1 of Firefox was working on all platforms.

The newest "Firefox" browser in Android is still 142.0.1 and is working. The newest "Firefox Focus" browser in Android is 143.0 and is therefore not working.

**Additional tests in Wokwi:**
It happens with the normal potentiometer and the slide potentiometer.
The sliders of the settings (for example of the MPU-6050) during runtime work fine.

urish, 2025-09-18:
Thanks for reporting, issue reproduced. Will investigate.

urish, 2025-09-18:
Looks like something is broken in the code that calculates the transform for the knob of the potentiometer, it gets a NaN value:

[image]

urish, 2025-09-18:
Fix is up.

closed 2025-09-18 by urish